start-jsk/denso #62: moveit "ABORTED: Solution found but controller failed during execution"

130s, 2015-11-03:
Random move with RRTConnectkConfigDefault planner do generate a plan but doesn't execute with the error message in the title.

ROS Indigo, vs060 1.1.0, open_industrial_ros_controllers 1.1.0

```
********************************************************
* MoveGroup using: 
*     - CartesianPathService
*     - ExecutePathService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

1446586737.881460857 INFO [/tmp/buildd/ros-indigo-moveit-ros-move-group-0.6.5-0trusty-20151015-0010/src/move_group_context.cpp:82(MoveGroupContext::status) [topics: /rosout, /move_group/planning_scene_monitor/parameter_descriptions, /move_group/planning_scene_monitor/parameter_updates, /move_group/monitored_planning_scene, /move_group/ompl/parameter_descriptions, /move_group/ompl/parameter_updates, /move_group/display_planned_path, /move_group/display_contacts, /move_group/trajectory_execution/parameter_descriptions, /move_group/trajectory_execution/parameter_updates, /move_group/plan_execution/parameter_descriptions, /move_group/plan_execution/parameter_updates, /move_group/sense_for_plan/parameter_descriptions, /move_group/sense_for_plan/parameter_updates, /move_group/result, /move_group/feedback, /move_group/status, /pickup/result, /pickup/feedback, /pickup/status, /place/result, /place/feedback, /place/status] MoveGroup context using planning plugin ompl_interface/OMPLPlanner
1446586737.881517193 INFO [/tmp/buildd/ros-indigo-moveit-ros-move-group-0.6.5-0trusty-20151015-0010/src/move_group_context.cpp:83(MoveGroupContext::status) [topics: /rosout, /move_group/planning_scene_monitor/parameter_descriptions, /move_group/planning_scene_monitor/parameter_updates, /move_group/monitored_planning_scene, /move_group/ompl/parameter_descriptions, /move_group/ompl/parameter_updates, /move_group/display_planned_path, /move_group/display_contacts, /move_group/trajectory_execution/parameter_descriptions, /move_group/trajectory_execution/parameter_updates, /move_group/plan_execution/parameter_descriptions, /move_group/plan_execution/parameter_updates, /move_group/sense_for_plan/parameter_descriptions, /move_group/sense_for_plan/parameter_updates, /move_group/result, /move_group/feedback, /move_group/status, /pickup/result, /pickup/feedback, /pickup/status, /place/result, /place/feedback, /place/status] MoveGroup context initialization complete
1446586738.577635252 INFO [/tmp/buildd/ros-indigo-moveit-ros-planning-interface-0.6.5-0trusty-20151015-0021/move_group_interface/src/move_group.cpp:150(planning_interface::MoveGroup::MoveGroupImpl::MoveGroupImpl) [topics: /rosout, /planning_scene, /planning_scene_world, /rviz/motionplanning_planning_scene_monitor/parameter_descriptions, /rviz/motionplanning_planning_scene_monitor/parameter_updates, /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update_full, /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update, /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/feedback, /trajectory_execution_event, /attached_collision_object, /move_group/goal, /move_group/cancel, /pickup/goal, /pickup/cancel, /place/goal, /place/cancel] Ready to take MoveGroup commands for group manipulator.
1446586738.577711271 INFO [/tmp/buildd/ros-indigo-moveit-ros-planning-interface-0.6.5-0trusty-20151015-0021/move_group_interface/src/move_group.cpp:804(planning_interface::MoveGroup::MoveGroupImpl::allowLooking) [topics: /rosout, /planning_scene, /planning_scene_world, /rviz/motionplanning_planning_scene_monitor/parameter_descriptions, /rviz/motionplanning_planning_scene_monitor/parameter_updates, /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update_full, /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update, /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/feedback, /trajectory_execution_event, /attached_collision_object, /move_group/goal, /move_group/cancel, /pickup/goal, /pickup/cancel, /place/goal, /place/cancel] Looking around: no
1446586738.577771422 INFO [/tmp/buildd/ros-indigo-moveit-ros-planning-interface-0.6.5-0trusty-20151015-0021/move_group_interface/src/move_group.cpp:810(planning_interface::MoveGroup::MoveGroupImpl::allowReplanning) [topics: /rosout, /planning_scene, /planning_scene_world, /rviz/motionplanning_planning_scene_monitor/parameter_descriptions, /rviz/motionplanning_planning_scene_monitor/parameter_updates, /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update_full, /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update, /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/feedback, /trajectory_execution_event, /attached_collision_object, /move_group/goal, /move_group/cancel, /pickup/goal, /pickup/cancel, /place/goal, /place/cancel] Replanning: no
1446586774.377104476 INFO [/tmp/buildd/ros-indigo-moveit-ros-move-group-0.6.5-0trusty-20151015-0010/src/default_capabilities/move_action_capability.cpp:93(MoveGroupMoveAction::executeMoveCallback_PlanAndExecute) [topics: /rosout, /move_group/planning_scene_monitor/parameter_descriptions, /move_group/planning_scene_monitor/parameter_updates, /move_group/monitored_planning_scene, /move_group/ompl/parameter_descriptions, /move_group/ompl/parameter_updates, /move_group/display_planned_path, /move_group/display_contacts, /move_group/trajectory_execution/parameter_descriptions, /move_group/trajectory_execution/parameter_updates, /move_group/plan_execution/parameter_descriptions, /move_group/plan_execution/parameter_updates, /move_group/sense_for_plan/parameter_descriptions, /move_group/sense_for_plan/parameter_updates, /move_group/result, /move_group/feedback, /move_group/status, /pickup/result, /pickup/feedback, /pickup/status, /place/result, /place/feedback, /place/status] Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
1446586774.377495398 WARN [/tmp/buildd/ros-indigo-moveit-ros-move-group-0.6.5-0trusty-20151015-0010/src/move_group_capability.cpp:91(MotionPlanRequest move_group::MoveGroupCapability::clearRequestStartState) [topics: /rosout, /move_group/planning_scene_monitor/parameter_descriptions, /move_group/planning_scene_monitor/parameter_updates, /move_group/monitored_planning_scene, /move_group/ompl/parameter_descriptions, /move_group/ompl/parameter_updates, /move_group/display_planned_path, /move_group/display_contacts, /move_group/trajectory_execution/parameter_descriptions, /move_group/trajectory_execution/parameter_updates, /move_group/plan_execution/parameter_descriptions, /move_group/plan_execution/parameter_updates, /move_group/sense_for_plan/parameter_descriptions, /move_group/sense_for_plan/parameter_updates, /move_group/result, /move_group/feedback, /move_group/status, /pickup/result, /pickup/feedback, /pickup/status, /place/result, /place/feedback, /place/status] Execution of motions should always start at the robot's current state. Ignoring the state supplied as start state in the motion planning request
1446586774.377640376 INFO [/tmp/buildd/ros-indigo-moveit-ros-planning-0.6.5-0trusty-20151014-1656/plan_execution/src/plan_execution.cpp:181(PlanExecution::planAndExecuteHelper) [topics: /rosout, /move_group/planning_scene_monitor/parameter_descriptions, /move_group/planning_scene_monitor/parameter_updates, /move_group/monitored_planning_scene, /move_group/ompl/parameter_descriptions, /move_group/ompl/parameter_updates, /move_group/display_planned_path, /move_group/display_contacts, /move_group/trajectory_execution/parameter_descriptions, /move_group/trajectory_execution/parameter_updates, /move_group/plan_execution/parameter_descriptions, /move_group/plan_execution/parameter_updates, /move_group/sense_for_plan/parameter_descriptions, /move_group/sense_for_plan/parameter_updates, /move_group/result, /move_group/feedback, /move_group/status, /pickup/result, /pickup/feedback, /pickup/status, /place/result, /place/feedback, /place/status] Planning attempt 1 of at most 1
1446586774.382498981 INFO [/tmp/buildd/ros-indigo-moveit-planners-ompl-0.6.7-0trusty-20151015-0011/ompl_interface/src/model_based_planning_context.cpp:234() [topics: /rosout, /move_group/planning_scene_monitor/paramet [... cut ...] _updates, /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update_full, /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update, /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/feedback, /trajectory_execution_event, /attached_collision_object, /move_group/goal, /move_group/cancel, /pickup/goal, /pickup/cancel, /place/goal, /place/cancel] ABORTED: Solution found but controller failed during execution
```

130s, 2015-11-03:
Rolling back to the last commit before the big changes introduced recently as following, controller started working fine. 

- open_industrial_ros_controllers to [this commit](https://github.com/start-jsk/open_industrial_ros_controllers/commit/4160d298af38535b5c5fd4cea37c8022bdb3e210) (Dec 13, 2013)
- denso to [this commit](https://github.com/start-jsk/denso/commit/5c23c6a2744bf54fdacd67f3b11f7550bf2a49ad) (Oct 8, 2015)

ompugao, 2015-11-04:
#64 will fix this issue

closed 2015-11-04 by 130s

130s, 2015-11-04:
And DEB release request is on board https://github.com/ros/rosdistro/pull/9763.

PR #64 (2015-11-04): fix dryrun mode
this pull-req will fix #62