start-jsk/denso #68: follow naming conventions

ompugao, 2015-11-15:
at some time...

I cannot send proper msgs to controller_manager and applications in the upper layer.

ompugao, 2017-02-03:
#82

k-okada, 2017-10-22:
@ompugao 

I have problem on launching new controller, 
```
[ WARN] [1508641039.999362668]: urdf limits int joint base_link_joint is not defined
[ WARN] [1508641039.999413780]: urdf soft limits int joint base_link_joint is not defined
terminate called after throwing an instance of 'joint_limits_interface::JointLimitsInterfaceException'
  what():  Cannot enforce limits for joint 'base_link_joint'. It has no velocity limits specification.
[arm_controller/denso_ros_control-3] process has died [pid 13761, exit code -6, cmd /home/k-okada/catkin_ws/ws_denso/devel/lib/denso_ros_control/denso_ros_control __name:=denso_ros_control __log:=/home/k-okada/.ros/log/b7ff1320-b6d4-11e7-8c29-1c4d70ae957f/arm_controller-denso_ros_control-3.log].
log file: /home/k-okada/.ros/log/b7ff1320-b6d4-11e7-8c29-1c4d70ae957f/arm_controller-denso_ros_control-3*.log
[INFO] [1508641040.611771]: Con
```
how did you fix that?
changed urdf ?
fixed ros_control ? https://github.com/ros-controls/ros_control/pull/221/files
remove fixed joint from controller?
```
    appendJoint(root_segment->second.children[0], joints);
    // skip fixed joint https://stackoverflow.com/questions/4478636/stdremove-if-lambda-not-removing-anything-from-the-collection                                          
    auto new_end = std::remove_if(joints.begin(), joints.end(),
                                  [](const KDL::Joint*& joint)->bool
                                  { return joint->getType() == KDL::Joint::None; });
    joints.erase(new_end, joints.end());
```