start-jsk/denso #78: Reduce build_depends of denso_launch

tfoote, 2016-06-23:
This is a launch package, that only installs resources. https://github.com/start-jsk/denso/blob/indigo-devel/denso_launch/CMakeLists.txt#L13

I suspect the build depends should be removable. This will save a lot of resources when building the package. On arm platforms it takes over an hour sometimes due to installing all these dependencies, while the actual compile of the package takes, a few minutes only. 

http://build.ros.org/view/Ibin_arm_uThf/job/Ibin_arm_uThf__denso_launch__ubuntu_trusty_armhf__binary/buildTimeTrend

A quick look suggests these dependencies are removable.
https://github.com/start-jsk/denso/blob/indigo-devel/denso_launch/CMakeLists.txt#L8
https://github.com/start-jsk/denso/blob/indigo-devel/denso_launch/package.xml#L14-L17

since the only content in the CMakeLists.txt is the install rule.

k-okada, 2016-06-23:
thank you, I have fixed this at #79

130s, 2016-07-14:
Fixed via #79

closed 2016-07-14 by 130s