start-jsk/denso #81: i cannot find the open_controllers_interface/open_controllers_interface.h

hzwuqijun, 2017-01-26:
when i was running the denso_controller/main.cpp, i cannot find the open_controllers_interface/open_controllers_interface.h in the system file,how can i do

k-okada, 2017-01-27:
that should be within ros-indigo-open-controllers-interface package, hope
this helps.
```
$ dpkg -L ros-indigo-open-controllers-interface
/.
/opt
/opt/ros
/opt/ros/indigo
/opt/ros/indigo/include
/opt/ros/indigo/include/open_controllers_interface
/opt/ros/indigo/include/open_controllers_interface/open_controllers_interface.h
/opt/ros/indigo/lib
/opt/ros/indigo/lib/libopen_controllers_interface.so
/opt/ros/indigo/lib/pkgconfig
/opt/ros/indigo/lib/pkgconfig/open_controllers_interface.pc
/opt/ros/indigo/share
/opt/ros/indigo/share/open_controllers_interface
/opt/ros/indigo/share/open_controllers_interface/cmake
/opt/ros/indigo/share/open_controllers_interface/cmake/open_controllers_interfaceConfig.cmake
/opt/ros/indigo/share/open_controllers_interface/cmake/open_controllers_interfaceConfig-version.cmake
/opt/ros/indigo/share/open_controllers_interface/package.xml
/usr
/usr/share
/usr/share/doc
/usr/share/doc/ros-indigo-open-controllers-interface
/usr/share/doc/ros-indigo-open-controllers-interface/changelog.Debian.gz
```

hzwuqijun, 2017-02-04:
Thanks for your answer.If I want to run this package on the real robot vs060，what file should I run？I want let the real vs060 do the path planning...

closed 2017-02-07 by hzwuqijun