start-jsk/denso #83: run on the real robot

hzwuqijun, 2017-02-07:
If I want to run this package on the real robot vs060，what file should I run？I want let the real vs060 do the path planning...

k-okada, 2017-02-08:
please try
https://github.com/start-jsk/denso/blob/indigo-devel/denso_launch/launch/denso_vs060_moveit_demo.launch

sato235, 2018-02-06:
Dear @k-okada and @hzwuqijun 
I can move the real VS-068 via rViz by "bringup" package.
And, I want to control real VS-068 by inputting joint angle data (prepared as json file).
I am planning to use "denso_robot_bringup" package and adding node "realtime_loop" and "send angle" nodes by python file modified from code shown on  chapter 3 (http://wiki.ros.org/open_industrial_ros_controllers/Tutorials/To%20program%20robot%20with%20open_industrial_ros_controllers)
However , there are many errors.

Which package or node should I use? "denso_robot_core?" "denso_controller"?