start-jsk/denso #89: URDF file for VS6577-BW*

Booooooosh, 2017-11-13:
Is there any URDF for VS6577-BW* model?

k-okada, 2017-11-14:
Unfortunately, we do not have URDF which is not listed in this repository. We need to find from somewhere.

Booooooosh, 2017-11-18:
Hi, many thanks for the reply.
But where can I find the URDF for this model? Or do I have to create one?
Many thanks!

Booooooosh, 2017-11-21:
Hi, just a follow up question. I looked through the urdf file for VS060. Just curious whether is it possible to replace the VS060 URDF with VS6577-BW URDF(I can try building the URDF myself, since I'm not using GAZEBO, it could be simpler), and keep the rest pf the launch file the same?
Will this work? Basically what I mean is that does the denso_ros_controller works for VS6577 also? (Since they are all 6DOF, I assume the controller can do the same thing for VS6577)?

Many thanks!

k-okada, 2017-11-23:
I think that should work

Booooooosh, 2017-11-23:
Alright, I’ve done the URDF for the VS6577, will make a try!
Many thanks!

发自我的 iPhone

Booooooosh, 2017-11-23:
> You can use ROSConverter.exe which is included in the latest WINCAPS III.
> The ROSConverter is located in the WINCAPS III install media:
> $(install media)\Unsupported\Tools\ROSConverter\Bin\ROSConverter.exe
>
> This tool convert WINCAP III project to ROS URDF.

Many thx! Though I believe I have an older version of WINCAPS. However, I have built the URDF, and can use it in the MoveIt, next step is to make sure that the controller can connnect to the real robot. By that I mean the vs060 ros controller can also be used for vs6557.

发自我的 iPhone

closed 2017-11-26 by Booooooosh