start-jsk/denso #91: ResourceNotFound: denso_controller on Kinetic

7675t, 2017-11-15:
Hi,

I tried sample launch file as:

```
roslaunch denso_launch denso_vs060_moveit_demo_simulation.launch 
```

results:
```
...
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: denso_controller
```

How can I fix this? Thanks.

k-okada, 2017-11-15:
That should be fixed in latest release, 2.0.2

So, if you’re not using shadow fixed, please use source tree.

http://repositories.ros.org/status_page/ros_kinetic_default.html?q=denso

7675t, 2017-11-15:
Thank you, I can just wait.

closed 2017-11-15 by 7675t