start-jsk/denso #97: arm.get_current_pose() results in ERROR: "Failed to fetch current robot state"

FlurinArner, 2017-12-12:
## Description
When requesting the current pose from the MoveGroupCommander("manipulator") through the [get_current_pose() Function](http://docs.ros.org/kinetic/api/moveit_commander/html/classmoveit__commander_1_1move__group_1_1MoveGroupCommander.html#a7afb57a5c62e9eb6abe5165421223ef1), the function fails.
At first I thought it was related to an [issue with moveit_planner](https://github.com/ros-planning/moveit/issues/579#issue-248818581), but if I build their source for moveit, the "Found empty JointState message" disappears, while this issue remains.
As soon as ROS Answers is back up, I'm going to post a question there and link it on&to this page.

## Your Environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source or Binary build?: Binary build for ROS, then source build for denso:kinetic-devel with commit: 3e725b1

## Steps to reproduce
* Create and move to workspace source folder:
    `mkdir -p denso_kinetic_devel_ws/src`
    `cd denso_kinetic_devel_ws/src`
* Download source for denso:kinetic-devel:
    `git clone git@github.com:start-jsk/denso.git`
* Edit following python script with preferred editor: 
    `denso/vs060/scripts/irex_demo_noteaching.py`
* Add in lines (after line 94: `arm.go() or arm.go() or rospy.logerr("arm.go fails")` ):
    `current_pose = arm.get_current_pose()`
    `rospy.loginfo("Current pose: \n{}".format(current_pose.pose))`
* Save file.
* Build workspace:
    `cd ..`
    `catkin_make`
* Source workspace:
    `source devel/setup.bash`
* Run launchfile:
    `roslaunch vs060_moveit_config demo_simulation_cage.launch`

## Expected behavior
arm.get_current_pose() returns the current pose of the end_effector (Flange in denso's case)

## Actual behavior
Error message: 
> [ERROR] [1513090715.277469722]: Failed to fetch current robot state

## Console output
[logfile_for_issue_2.txt](https://github.com/start-jsk/denso/files/1552311/logfile_for_issue_2.txt)
(I only removed username@computername references)

FlurinArner, 2017-12-13:
Following [this answer](https://answers.ros.org/question/252114/failed-to-validate-trajectory-couldnt-receive-full-current-joint-state-within-1s-error/?answer=252620#post-id-252620) was finally able to solve the problem by editing the launchfile:
* In `denso/vs060_moveit_config/launch/demo_simulation_cage.launch`, after line 8:
`<node name="spawn_cage" pkg="vs060" type="irex_demo_noteaching.py" />`

* Add the following lines:
`  <node pkg="joint_state_publisher" type="joint_state_publisher" name="joint_state_publisher" output="screen">`
`    <param name="publish_frequency" type="double" value="30.0" />`
`    <rosparam param="/source_list"> [/arm_controller/joint_states] </rosparam>`
`  </node>`

Now the current_pose is returned and all works, but this feels much like a hack around the problem.
I doesn't make sense to publish the same information for joint_states, under two separate topics (`/joint_states` and `/arm_controller/joint_states`).
Maybe this is more of a problem with moveit_commander, than with denso..
Feel free to close this "Issue", and I'll be asking the same question on answers.ros.org

Heisenberg1997, 2018-03-28:
can you give a more detail Introduction about that. I have the same problem and I don't know the which joint pub need change. please