start-jsk/denso #100: How can I make a package for DENSO robot (other than vs060)?

Landau0323, 2018-06-04:
**What I want to do**: 
1. I want to give vs068 the values of its angles (J1 - J6) as series of time, which give a single orbit.
2. After that, I want to move this robot on rviz, and get the coordinate (x,y,z,orientation) of its J6 as series of time (namely, I want to perform the forward kinematics).

**Current status**: By using your package, "denso", I could yield an orbit of vs060 on rviz.
(To be more specific, I used "denso_vs060_moveit_demo_simulation.launch")
However, I do not know how to do the same thing for vs068.

**Question**: How can I make a launch file which has the same function as "denso_vs060_moveit_demo_simulation.launch", but for vs068 (not for vs060)?

I have gone through this and the other launch files in the package, but I was overwhelmed and maybe was too lazy, so that I could not find which part of the launch file should be changed in which way.
Most the related files seem to be located in the directory "/opt/ros/indigo/share/vs060_moveit_config/launch", so I guess that I should make the vs068-version of this directory, but I do not know how...
If anyone tells me how to do this, I would be very grateful.

For reference, my environment is Ubuntu 14.04, ROS indigo.
I already have "vs068_description", which is a model file for vs068, given by "denso_robot_ros/denso_robot_descriptions".

Moirai, 2018-06-06:
* How did you make a vs068_moveit_config ?
* Did you make own workspace?

YoshihiroMIYAKOSHI, 2018-06-09:
The denso_robot_ros package is a newer version of denso package.
So if you can migrate to the denso_robot_ros package, you can easily control vs068.
Please refer to the [denso_robot_ros's wiki](http://wiki.ros.org/denso_robot_ros).

closed 2018-06-12 by Landau0323